magicbug/Cloudlog #332: statictics custom seems to be missing directory variable

janvgils, 2019-07-28:
**Describe the bug**

When selecting <cloudlog>/index.php/statistics custom the <cloudlog> part is missing and falls back to the webserver root. For general and satellites it is working fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'statistics'
2. Click on 'custom'

**Expected behavior**
This selection should bring you to the custom statistics function and not to the webroot.

**Screenshots**
A screen shot is difficult because ones needs to try a catch a mouse over.

**Desktop (please complete the following information):**
 - OS: [e.g. Linux Debian]
 - Browser [e.g. Firefox]
 - Version [e.g. 60.8.0esr x64]

**Additional context**
Add any other context about the problem here.

Regards, Jan PE0SAT

janvgils, 2019-07-30:
I saw some part where not visible in the original post \<cloudlog\> so changed to issue information.

magicbug, 2019-09-12:
Yes, removed as it wasn't working, it will be replaced my /search/filter which is now in beta if you go to the url with the latest codebase.

closed 2019-09-12 by magicbug