magicbug/Cloudlog #335: Au

magicbug, 2019-08-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ea5wa, 2019-10-31:
Why not a combobox with satellite names, selecting one of them automatic population of sat mode, Mode, Frequencies.
It would be perfect solution for Satellite Portable operation

ok1cdj, 2020-04-17:
Please use this ARRL list to have it correct for LOTW upload.

magicbug, 2020-05-17:
> Please use this ARRL list to have it correct for LOTW upload.
> https://lotw.arrl.org/lotw-help/frequently-asked-questions/#sats

I'd recommend using the actual satellite names, it can then be adjusted for LOTW because satellite applications don't use LOTW names.

closed 2020-05-21 by magicbug