magicbug/Cloudlog #336: Have a use for the field OPERATOR

phl0, 2019-08-27:
We are evaluating Cloudlog for our clubstation currently. It would be useful to have an information about who (personally) operated the club callsign for each QSO. From what I read in the ADIF specs there is a field OPERATOR which should allow for that.
We imported an .adif file and the field obviously gets filled. So what is missing is a way to have that on the input side. So QSOs which are logged through the Cloudlog GUI (aka. web interface) do not populate this field.
Or would that require separate accounts for anyone who wants to log QSOs for the club station?

magicbug, 2019-08-27:
@phl0 In the code it uses the user logins callsign against the OPERATOR field so if you give everyone an account, at the moment the privileges don't really do anything so I'd set everyone to admin for now.

I need to expose the OPERATOR field on the QSO popup windows so that it can be seen.

What I'm planning todo do is add a role that removes the admin features and just lets an "Operator" add QSOs.

magicbug, 2019-08-27:
Station Callsign is for argument's sake in Cloudlog the logbook callsign so 2M0SQL this is defined in Station Profiles, then you can select it in the QSO panel "general"

magicbug, 2019-08-27:
Yes, I guess a global club username for everyone to use is an option, i just never saw it like that.

magicbug, 2019-08-27:
I'll get it added :)

magicbug, 2019-08-27:
Done

magicbug, 2019-08-27:
When you create an account for a user, when you fill in "Callsign" this will be the field that is used for the OPERATOR field

phl0, 2019-08-28:
I see. But how is station callsign set then?!

phl0, 2019-08-28:
I see. Means we need to have an account for anyone who wants to log for our club call sign. So then change this to FR: Make the COL_OPERATOR filed visible :)

phl0, 2019-08-28:
Not a problem. I would find a separate account for each operator more convenient.
But I just discovered, that OPERATOR is also not included when exporting. Could you please also consider this?

phl0, 2019-08-28:
Mni tnx.

phl0, 2019-08-28:
Cool. Will test later on as I am horizontally polarized :-)

phl0, 2019-08-28:
Am I missing something on the QSO details popup window?

[image]

Should include the operator as per 47d9efe218986142bd8af1c4d5d78b0db8f19eac right?

phl0, 2019-08-28:
Well, forget what I worte. I didn't realize there are veritcal scrollbars ... *facepalm*

magicbug, 2019-10-05:
This is now 100% stored from User account and won't be made available in the Panel unless under edit to correct an error.

closed 2019-10-05 by magicbug