magicbug/Cloudlog #338: Export to eQSL does not respect "ignore"-flag in EQSL_QSL_SENT

dg9vh, 2019-08-29:
**Describe the bug**
Export to eQSL does not respect "ignore"-flag in EQSL_QSL_SENT

**To Reproduce**
Steps to reproduce the behavior:
1. Edit QSO
2. Set eQSL-Sent state to Invalid (Ignore)
3. Export eQSL
4. See error

**Expected behavior**
We should add the Invalid (Ignore) state into the list that is not exported to eQSL. I think, it would be done in the Logbook_model.php at line 929:
      $this->db->where('COL_EQSL_QSL_SENT  !=', 'Y');
This line has to be changed to an "not in" (where COL_EQSL_SQL_SENT not in ('Y', 'I')).

magicbug, 2019-08-29:
This seems a reasonable request and I agree should be set to ignored if thats the option selected.

closed 2019-08-30 by magicbug

dg9vh, 2019-09-01:
Fixed with https://github.com/magicbug/Cloudlog/pull/340

commit 09a46b2b (2019-08-30): Fixes #338 Respects the Ignore flag when exporting to eQSL