magicbug/Cloudlog #342: Allow Date Selection on ADIF Export

magicbug, 2019-09-04:
**Is your feature request related to a problem? Please describe.**

Allow users to export ADIF records by defining a date period

**Describe the solution you'd like**

Input fields for start and end date that have a calendar popup that allows you to define the two data points followed by an export button.

ea5wa, 2019-10-14:
It could be very convenient at the end of a field day

AndreasK79, 2020-02-07:
I have a working concept with minor adjustments to 2 files. It works, and most new browsers have a datepicker built in. A bit more work is needed to add a javascript bootstrap datepicker. Any suggestions, or am I free to pick one?

[image]
[image]

magicbug, 2020-02-07:
As long as it works with bootstrap looks ok on desktop and mobile no real constraints.

AndreasK79, 2020-02-09:
I guess this can be closed now with the commit I made two days ago?

closed 2020-02-09 by magicbug