magicbug/Cloudlog #344: Query QRZ.com

myzinsky, 2019-09-05:
It would make sense to retrieve OM's information from QRZ.com. Of course a QRZ.com username and password must be provided. Some inspiration can be found here: https://github.com/zebpalmer/pyQRZ

dg9vh, 2019-09-05:
Cloudlog already has this functionality built in when logging a qso. Requirements are a valid qrz.com-login.

73 de Kim
DG9VH

magicbug, 2019-09-05:
Please consult the wiki callsign lookup as been integrated with Cloudlog since the first release  [https://github.com/magicbug/Cloudlog/wiki/Callsign-Lookup](https://github.com/magicbug/Cloudlog/wiki/Callsign-Lookup)

closed 2019-09-05 by magicbug

oe3ide, 2020-03-05:
Hi,
I followed the instructions:
https://github.com/magicbug/Cloudlog/wiki/Callsign-Lookup

How can I check, that the lookup is working? If I enter a valid Call and press TAB, nothing happens (ok, lookup of previous QSOs is working). But no infos from QRZ.com.
User/password is 100% correkt - double-checked.

Thanks & 73s