magicbug/Cloudlog #346: FR: Take Image & Save Paper QSL Card Graphics (Low Priority)

magicbug, 2019-09-05:
**Describe the solution you'd like**

The interface that allows taking a photo of a paper QSL card front and back with a cropping interface which is then saved in /assets/qslcards/ this is then made available in the QSO popup windows or in a dedicated area for storage in case of losing it.

This interface would be designed to work on mobile phones, no need to scan, just snap a photo front and back.

**Describe alternatives you've considered**

Keeping cards in a shoebox.

M0TZO, 2020-02-22:
_I was going add new suggestion, but saw this. My idea got this far before I saw other post_.:-

I’d  like to be able to upload front and back of QSL cards (I’ve not got any, but seen designs with insides too.

It may also be useful to have a button to mark as private / public depending on initial preference so that ones with personal info can be hidden, but viewable to admins.

AndreasK79, 2020-10-28:
So I thought this would be a fun challenge to implement. I have it working, just need a little bit of tweaking before I can create a pull request. 

Here are some previews:
[image]
[image]

I have only implemented a file upload, but it works with a mobile phone. No cropping interface, as I haven't found a good solution for that. I figure this might be a good start, and the functionality can be expanded.

M0TZO, 2020-10-28:
For cropping, most phones can do it before uploading, and I like using the Google Photoscan app as automatically scans and crops.

magicbug, 2020-10-28:
Looks a great start, totally agree, fancy functions can be added further down the development 👍🏻

AndreasK79, 2020-10-28:
@M0TZO Ok, perhaps I misunderstood things a little bit then. I thought about cropping in the browser, not in the phone itself.

@magicbug Well then, I'll tweak things a bit before I submit the code. Things are working now, but I need to test a bit more.

magicbug, 2020-10-28:
Fab, ill be back to coding soon, been battling computer issues this week which is ripped into my free time :( (one word drivers hell!)

M0TZO, 2020-10-28:
One more thought, some QSL cards are 4 sided, I have at least one of those, so may need (even later) tabs for those. Look forward to testing this soon.

AndreasK79, 2020-10-28:
Perhaps I need to rethink things a bit. A carousel might be an option. Something like this: https://getbootstrap.com/docs/4.0/components/carousel/

That way, people can add as many pictures as they want.

AndreasK79, 2020-10-28:
PR #677 has been submitted. I think that should be it for a first release of the function. Carousel implemented, which simplified things a lot. Now you can add as many QSL images as you want.

AndreasK79, 2020-10-29:
Here is a preview of the QSL management:
[image]

You need to be logged in to view and manage QSL cards.

The QSL Card tab is a carousel looping through the images.

A new page was added under QSO -> View QSL. This page contains all the QSL cards, where you can delete and view the cards.

closed 2020-11-01 by magicbug