magicbug/Cloudlog #347: Contest mode

M0LTE, 2019-09-07:
**Is your feature request related to a problem? Please describe.**
When I'm operating a contest it would be nice to have an interface which facilitated contesting.

**Describe the solution you'd like**
Boxes for sent and received exchange (serial), optional auto increment sent serial.
This might be on a third screen under the QSO menu.

**Describe alternatives you've considered**
Considered building a separate app that uses the API to post QSOs in to cloudlog. But would be best integrated.

magicbug, 2019-09-08:
Hi Tom,

Agree, I think this would be better as its own "screen" interestingly if you go back to very early codebase you'll find some contesting bits but they never got moved over instead i binned them.

Think a screen to mimic N1MM+ QSO box could be a reasonable start.

M0LTE, 2019-09-08:
Thumbs up from me.

ok1cdj, 2020-04-16:
Look here: https://remoteqth.com/wiki/index.php?page=PHP+contest+Log
I thing this is enough for cloud based contest log and you can do it it same way.

magicbug, 2020-04-16:
[image]

Yes think the RemoteQTH style fields for contesting is most appropriate, not including score counting but just having a field for doing serial numbers etc I think would be reasonable basic for most hams just giving away points.

I actually want to sit down and redesign the QSO area just need time which is always the problem!

AndreasK79, 2020-11-16:
Ok, so I started playing with things again :)
Here is my take. It is not done, and it is not logging right now. Much of the form is done. I'm implementing different things in javascript, like keyboard shortcuts, jumping to input fields and incrementing serial when logging.

When logging, the table at the bottom is filled, and I also implemented the callsign search from the original logging, but with a highlighting feature, 

The biggest task left, is to get the logging working as it shoudl
Here is a preview:
[image]

@magicbug I might need some help with the logging in the back-end, but we'll see how this comes along.

M0LTE, 2020-11-16:
I like this and I’m interested. I don’t like having to drop to N1MM+ for
contests. Have you considered contests with differing exchanges? You will
note that N1MM+ has got an awful lot of logic around this area. Don’t let
‘perfect’ be the enemy of ‘working’ though!

Cheers and following with interest.

AndreasK79, 2020-11-17:
@M0LTE  I guess this version can be pushed to the code when the back-end logging is working.

For now, I've implemented CTRL-Enter for logging. CTRL-W for wipe, and space in the callsign-field and exchange received-field to jump between those two.

As for the exchange, I had in mind to have serial and other. If you look at the picture, you can see exchange type.
If it's serial, it will be incremented with 1 when logging. If it's other, it will not be wiped when logging.

AndreasK79, 2020-11-17:
#708 is added now. Basic stuff is in place.

magicbug, 2020-11-17:
This is looking great, happy to look over the saving of the QSO, not a problem, like you say think it just needs basic functionality, not a N1MM replacement as such. I'll give the PR a check over.

M0LTE, 2020-11-17:
Yep, those keys sound good!

AndreasK79, 2020-11-17:
ESC is also implemented, and does the same thing as CTRL-W. I noticed that was used in the regular QSO logging.

m0pcb, 2020-11-17:
I thought most contest loggers used Alt-W for 'wipe' rather than Ctrl-W.
Maybe it's both?

73, Iain AD5XI / M0PCB

magicbug, 2020-11-17:
Alt + W was always the standard in n1mm, wt & dxlog I believe browsers will let us use that combo too

AndreasK79, 2020-11-17:
I thought it was reserved, but that is an easy change in that case.

AndreasK79, 2020-11-18:
Forget about CTRL-W. I had implemented ALT-W. I don't know why I wrote CTRL-W.

AndreasK79, 2020-11-30:
@magicbug anything missing, or need to be changed? If not, I don't see why this can't be merged.

magicbug, 2020-12-07:
OK have some more time to just go through this and its going to just be bits off the top of my head so sorry

- I'm guessing this just uses the default station profile
- I'm guessing CAT radio functions arent there
- Frequency when saved is "zero" we do have some functions in the main QSO area to select the basic freq this will need similar function
- The logging session only is a bit annoying as I left the screen and lost my QSOs.. not sure the solution to this perhaps we can store a local session ID to match when first QSO was added and filter contest qsos based on the first qso saved not sure.
- Serial is great, not sure how other is meant to function? be nice to have a serial plus gridsquare option for the VHF+ contest dabbling
- When in Exch (R) it would be good if you can hit enter to save the QSO

magicbug, 2020-12-07:
Bit more 

- Keyboard shortcuts don't seem to do anything apart from ESC at least for me

AndreasK79, 2020-12-07:
Thanks for the feedback.

- It uses the default station profile
- CAT is not there. Not sure how that works, but I'll see if I can figure it out
- I'll take a look at the frequency
- I never gave the logging session much thought, other than seeing what you just logged. Need to figure out something.
- Other is for non-serial (CQZ for CQWW, Power for ARRL). When serial is chosen, the number is incremented by 1. 
- Enter can be implemented. Right now, CTRL-Enter is implemented for logging
- I implemented SPACE, ALT-W, ESC and CTRL-Enter. They work for me, so I don't know what is going on here.

magicbug, 2020-12-07:
I can do the CAT bits for you, I probably know that mess better than most.

I wonder if it extensions in Chrome, need to see if I can force Google Chrome to enable webpage ones!

AndreasK79, 2020-12-07:
I got CAT working by adding radio-select and including javascript from the qsopage:
[image]
I just need to figure out why it is no longer logging to the database. As soon as I figure it out, I'll make another commit you can try.

magicbug, 2020-12-07:
Look forward to more testing

AndreasK79, 2020-12-07:
Commit with some fixes was added. CAT is working and frequency is saved. Give it a go.

magicbug, 2020-12-09:
I'm happy with this release its got its quirks but we can work those out in beta I cant see anything that would hurt the other QSO entry areas I'm correct in thinking this @AndreasK79 if so safe to merge and we can move on to improving it?

Created a wiki page https://github.com/magicbug/Cloudlog/wiki/Contesting

magicbug, 2020-12-09:
@AndreasK79 also agree to try keeping as much of its js in separate files and views think this will be a good way of cleaning up footer.php is its getting a bit messy also as ever comments in code 👍🏻

AndreasK79, 2020-12-09:
@magicbug Yes, please merge and move this to beta. I do not see anything that should hurt the entry areas. Regular logging works, and I barely touched logbook_model. The only thing adjusted, was to add serial logging, but has a check for exchangeradio ( if($this->input->post('exchangeradio')) {).

I agree about the js part. Footer.php is too long. I moved everything I could to a separate js. Perhaps it's time to clean it up a bit more.

magicbug, 2020-12-09:
Get this back opened as we need it, yeah thought it was just that, but yeah footer.php is a bit insane and horrible hah

closed 2020-12-11 by magicbug