magicbug/Cloudlog #350: Support Date format YYYY-MM-DD as per ISO8601 spec for dates displayed within Cloudlog

magicbug, 2019-09-16:
I suspect some USA users have quietly hoped this would get addressed, make it an option on how to set the dates YYYY-MM-DD or the way UK do it DD-MM-YYYY :)

g0wfv, 2019-09-23:
To be honest, I'd like to see a date picker rather than have to manually input the date by figures every time!

I've just run into a bug on the log entry page whereby I entered the date "15th of June 2019" as 15-06-19 and it was input to the database as "19th of June 2015" (dates stated in plain English to avoid confusion!)  I guess it expects dd-mm-yyyy format else it assumes something else by using best guess?!

dg9vh, 2019-09-23:
I also agree, that a date picker (maybe in Combination with a text-box to enter a date in a defined format) would be the best.
It could be something we likely have at google's calendar for example:
[image]

magicbug, 2019-09-23:
I don't disagree, but that wasn't really what I was referring to. 
There's plenty libraries to use to address the date topup

dg9vh, 2019-09-23:
hm... This is a thing called i18n I think :-) The way could be to have an admin-panel where to set some localization info (for example miles vs. kilometers or currency or date/time-formats) and then use them as input-mask for the date-time-pickers...

Don't know, if codeigniter has something for this on their toolboxes :-) but I think this is not a new problem in the world, so there should be an easy to use solution for...

phl0, 2020-02-20:
+1 here for a user-configurable date format :)
As per https://twitter.com/flo_0_/status/1230540724749381635?s=20

magicbug, 2020-02-21:
As @phl0 poked me and reminded me I've made date format configurable with the latest commit, you can define whatever format you want inside /application/config/cloudlog.php and changing the variable inside `qso_date_format`

This currently doesn't change the date on the input box on the QSO page (Need to check the code behind that) but overall should help people.

closed 2020-02-21 by magicbug