magicbug/Cloudlog #351: Push QSOs to QRZ Logbook

magicbug, 2019-09-16:
It's been requested a few times and although not something I personally would use, Cloudlog should support pushing QSOs to QRZs Logbook service.

Integration actually looks straightforward just post it adif line, a lot of the Clublog code could actually be used. If anyones interested in helping?

Spec sheets here http://files.qrz.com/static/qrz/The%20QRZ%20Logbook%20API.pdf

magicbug, 2019-09-16:
Notes

- Galancing at the documentation, it will need a field in the station profile table for defining the API key for the logbook.

DJUnreal, 2019-10-06:
I can't help, but I'll +1 this as a feature request - at the moment my QRZ logbook gets populated from my LoTW logbook, so it's a pretty manual process. The more automated pushing of QSOs about, the better!

motoionic, 2019-12-17:
I was just coming over here to ask if you could add QRZ support.

magicbug, 2019-12-18:
The same answer stands, other priorities but anyone welcome to add it via a Pull Request.

AndreasK79, 2020-05-07:
Added #489 
@magicbug: 
Perhaps needs some more work. What should be done if upload fails?
Should we mark each qso as uploaded or not? Add another column in logbook-table?
Should a function for mass-upload be made?

URL should perhaps be moved to a table in the database.

magicbug, 2020-05-07:
Theres two fields in the main QSO table

- COL_QRZCOM_QSO_UPLOAD_DATE (the date the QSO was last uploaded to the QRZ.COM online service)
- COL_QRZCOM_QSO_UPLOAD_STATUS (the upload status of the QSO on the QRZ.COM online service)

Status can be Y/N/M see [https://www.adif.org/310/ADIF_310.htm#QSOUploadStatus_Enumeration](https://www.adif.org/310/ADIF_310.htm#QSOUploadStatus_Enumeration) this seems best way to handle if upload files just mark it as N

Also might be worth adding the M flag if a QSOs been edited so it gets reuploaded via the sync script?

AndreasK79, 2020-05-07:
So much for not checking the table for columns. I'll add some more stuff in.

AndreasK79, 2020-05-07:
@magicbug I made the modifications. Now all that is needed, is functionality for uploading everything that has not been uploaded before. At least basic support is in place.

magicbug, 2020-05-07:
OK, code looks fine, I've no real desire to use QRZ Logbook so I'll have to let others test its functionality to that degree, but thanks @AndreasK79 sure this will please people

magicbug, 2020-05-07:
@AndreasK79 Can I be annoying and ask if you could create a short "how to use it" on the Wiki for people

AndreasK79, 2020-05-08:
@magicbug I can do that. Not sure when I'll get around to it. In Cloudlog, all you need to do, is to save the QRZ logbook API KEY under station profiles. If that is found, the QSO will be uploaded when logging normally.

magicbug, 2020-05-08:
Question if QRZ is down will the system fail gracefully and QSOs will still get logged locally?  I say that as the servers are down a lot

AndreasK79, 2020-05-08:
It should. If API KEY exists, COL_QRZCOM_QSO_UPLOAD_STATUS is marked as N.
This value is only overwritten if RESULT=OK is returned.

This code returns false if error:
`$content = curl_exec($ch);
      if ($content){
          if (stristr($content,'RESULT=OK')) {
            return true;
          }
          else {
            return false;
          }
      }
      if(curl_errno($ch)){
          return false;
      }
      curl_close($ch);`

I tried yesterday to change the url, and the QSO was marked with N.

magicbug, 2020-05-08:
OK, created a page [https://github.com/magicbug/Cloudlog/wiki/QRZ-Logbook-Synchronisation](https://github.com/magicbug/Cloudlog/wiki/QRZ-Logbook-Synchronisation)

Is anything further needed for this feature? or can we officially close it

AndreasK79, 2020-05-08:
Thanks. Well, edited QSOs are not uploaded automatically. Only QSOs that are logged.
Not sure how to handle the edited QSOs. Should we just upload them again? If not, we need to get the id and store it away, then delete and reupload. 

Guess mass-upload needs to be added as well.

If we look past these two things, this can be closed.

magicbug, 2020-05-08:
I'd say handle editted via mass upload that could be run as a cronjob once a day or via a button in the interface maybe (sync button) in the station profile page

dg9vh, 2020-05-08:
Thats a good idea - like we do it with clublog... there is an API-call to call once a night via cronjob...

Those things are not time-critical I think so they can be handled with bulk-upload and have not to be done just-in-time.

magicbug, 2020-05-08:
Yep exactly, one of the reasons I never really made Clublog real-time its fairly non-important, it could wait an hour or two

dg9vh, 2020-05-08:
I update mine once a day at 0 UTC or something like that

AndreasK79, 2020-05-10:
So I've looked into doing some mass-upload. I tried to send several QSOs at once, but QRZ only seems to accept one at a time, so I'm not sure if it's a good idea to make one call to the server for each QSO.

magicbug, 2020-05-10:
If that's all it takes then yes reasonable to make multiple requests

magicbug, 2020-05-10:
Although it's weird because looking at the documentation it implies as long as it's seperated by EOR same as a adif file it can take more than one QSO

AndreasK79, 2020-05-10:
Yes, I read that too. I tried with several QSOs in one go, but only one showed up. I'll try one more time, if not working, I'll make a PR which uploads one QSO at a time.

AndreasK79, 2020-05-10:
No luck with several QSOs in one call. The question now is, how should the url be called? Should it be called with just a call, like the method used for Clublog?
Or should we go for call and grid?
Or should we just find all station_id's in the database with an API defined, and upload all QSOs?

magicbug, 2020-05-12:
I'd suggest people might want either

- Complete mass upload, if station profiles support HRD Log then to upload (likely to be a cron)
- Upload for each logbook within the UI

AndreasK79, 2020-05-12:
#494 added. Mass upload for cron job. Will make UI mass upload later.

magicbug, 2020-05-12:
@AndreasK79 perfect, I've added a bit to the wiki if you can make sure its correct to match how the code works. https://github.com/magicbug/Cloudlog/wiki/QRZ-Logbook-Synchronisation

AndreasK79, 2020-05-12:
@magicbug That's correct. When that url is called, each station_id with an API is fetched.
Then a loop is running and uploading all QSOs that are not uploaded.

AndreasK79, 2020-05-20:
#502 This should finalize this.

dgdavid, 2021-01-26:
@AndreasK79, 

Maybe it is obvious for most of you or maybe the API requirements just changed, but I think it worth to mention that such feature needs a QRZ susbscription. See https://www.qrz.com/docs/logbook/QRZLogbookAPI.html. Although they said that

> Some advanced features require a valid subscription while the majority of operations are free to all QRZ members

the reality it that all listed actions (INSERT, DELETE, UPDATE, FETCH) have the _"subscription required"_ note.

Did it work without subscription before? Am I doing something wrong? Did I miss-understand something?

Thanks!

magicbug, 2021-01-26:
I'll re open but might have suited a new issue.

I'd suggest we add an alert warning where the qrz API key is added.

dgdavid, 2021-01-26:
> I'll re open but might have suited a new issue.
> 

Sure. I didn't it because I was not sure if it is something new of it was a requirement at the time to implement it. That's why I decided to ask directly here. If it is better, I can open a new issue.

> I'd suggest we add an alert warning where the qrz API key is added.

Agree. I'd update the [_QRZLogbook_ wiki page](https://github.com/magicbug/Cloudlog/wiki/QRZ-Logbook-Synchronisation) too. 

PS: I just realized that it is mentioned in https://github.com/magicbug/Cloudlog/wiki/Callsign-Lookup

magicbug, 2021-01-27:
Note added to the wiki at the top, will do the UI in Cloudlog tomorrow.

AndreasK79, 2021-01-27:
@dgdavid Peter beat me to it. We are lacking documentation here and there, so no wonder you might be confused. Thanks for bringing it to our attention.

commit e5ad2630 (2021-01-27): [Station Profile] Added warning that QRZ Logbook requires paid subscription #351

This addresses comment that QRZ logbook needs a paid subscription to function fixes #351
closed 2021-01-27 by magicbug