magicbug/Cloudlog #353: Discussion: Exporting QSLs for printing based on active Station?

dg9vh, 2019-09-25:
Since we have now the possibility to create different station profiles and export adif files based on this, I think, it would be suitable to decide, how to export the "print lists". Here are different possibilities with pros/cons on them:

1. Exporting all QSOs resquested for printing (actual behavior) not looking on what station it is logged.
2. Exporting QSOs for active station (would be exporting each station separately by switching each station profile active)
3. Creating a list (checkboxes) of stations, that should be exported in one file based on the selection within the list (means, you have a list of all station-profiles with checkboxes to select and export).

The same would be for marking exported... 

Now it's time to consider and discuss about this...

dg9vh, 2019-09-27:
new PR https://github.com/magicbug/Cloudlog/pull/354 is creating the behavior of "Exporting QSOs for active station (would be exporting each station separately by switching each station profile active)"

magicbug, 2019-09-28:
Merged with master

closed 2019-10-03 by magicbug