magicbug/Cloudlog #356: New Install - parsing error when creating a Station Profile

WizGnome, 2019-10-03:
I have just installed Cloudlog on my web server. 
I have created a user account and updated Country Files
When I attempt to create a Station Profile (callsign EA7KIP) I get the following error:
Parse error: syntax error, unexpected 'list' (T_LIST), expecting identifier (T_STRING) in /var/www/Cloudlog/application/models/Dxcc.php on line 134
A PHP Error was encountered
Severity: Parsing Error
Message: syntax error, unexpected 'list' (T_LIST), expecting identifier (T_STRING)
Filename: models/Dxcc.php
Line Number: 134

magicbug, 2019-10-03:
Mike, when you updated the country files did it show "done" & also what version of php

WizGnome, 2019-10-03:
PHP is 5.6.39 - does it require a later version?

Yes, it did say "done" after updating the country files (I can see data in the mysql tables).

magicbug, 2019-10-03:
As per the readme requirements 

> PHP 7 recommended.

closed 2019-10-03 by magicbug

WizGnome, 2019-10-04:
The issue is that in php 5.6 'List' is a reserved word and is not allowed as a function name, changing list in Dok.php, Dxcc.ph and Station.php to mylist solved the issue (until I can upgrade my os to one which supports php 7)

magicbug, 2019-10-04:
Indeed, I'm aware why, but as PHP5.* has been end of life support now for too long decided it could stay.