magicbug/Cloudlog #357: Update DXCC stuck on downloading file

TheMcBain92, 2019-10-03:
I installed cloudlog and got it all up and running following the setup. When I click the 'update DXCC' button it comes up 'Downloading file.....' it then stays on this and doesn't change.

I have left this page open for a good hour or so and it never says that it is completed.

Stephen.

magicbug, 2019-10-03:
Have you made sure the /uploads is writeable as per the wiki install information?

TheMcBain92, 2019-10-03:
The folder should be writable. It is being hosted externally so only have cpanel access to set permissions.

magicbug, 2019-10-03:
My guess is that its something todo with that, or the servers stopping Clublog.org from being reachable which is where all the data comes from

TheMcBain92, 2019-10-04:
Thanks

I will do some more investigating

TheMcBain92, 2019-10-04:
Did a quick test today and I am able to create files inside the uploads folder using a php script so from that I can assume that It have write access to the folder.

TheMcBain92, 2019-10-04:
Issue solved. Turned out it was a php.ini setting issue in the top level folder. Even though 'allow_url_fopen' was set to enabled (set through cpanel) it did not filter through to the sub-directories of Cloudlog preventing the download from clublog.

Temporarry fix was to copy the php.ini to the sub-directories although a change to the .htaccess file will give the same result.

closed 2019-10-04 by TheMcBain92