magicbug/Cloudlog #363: ADIF Header Handling #bug

DJUnreal, 2019-10-06:
At the moment, ADIF imports don't seem to be able to handle header data. If an ADIF file with a header is uploaded, the upload page states the file type is invalid.

I would suggest having the ADIF parser begin by looking for an <EOH> tag (seems to be a standard header tag in ADIF), and ignore everything above that tag. This should then allow it to parse files correctly even with headers in there.

magicbug, 2019-10-06:
It actually handles header data fine, what it doesn't handle is free flow text thats become popular in ADIF files of late (no idea why)

DJUnreal, 2019-10-06:
Thanks Pete!

AndreasK79, 2020-02-05:
I had this problem when testing out the log as well. I traced it to the mime-check done by Codeigniter. Modifying mimes.php solved the problem for me.

'adi'   =>  array('application/octet-stream','text/plain', 'audio/x-hx-aac-adif'),
'adif'  =>  array('application/octet-stream','text/plain', 'audio/x-hx-aac-adif'),
'ADI'   =>  array('application/octet-stream','text/plain', 'audio/x-hx-aac-adif'),
'ADIF'  =>  array('application/octet-stream','text/plain', 'audio/x-hx-aac-adif'),

dg9vh, 2020-02-05:
How about a pull-request? @AndreasK79

AndreasK79, 2020-02-05:
Sure. I'll look into it later. I just need to get things set up.

magicbug, 2020-02-05:
Thanks, PR definitely would appreciated.

commit c682b8f3 (2020-02-05): This fixes the ADIF header import bug. Fixes #363.
commit da110b1f (2020-02-05): Merge pull request #408 from AndreasK79/master

This fixes the ADIF header import bug. Fixes #363.
closed 2020-02-05 by magicbug