magicbug/Cloudlog #364: Edit QSO Pop-up Not Formatting Correctly

DJUnreal, 2019-10-06:
When I try to edit a QSO, the popup window is incredibly small in a desktop web browser, and very misshapen on a mobile phone browser.

Suggest that instead of using size constraints, the pop-up could be allowed a percentage scale in each direction, scaling as far as it needs to in order to a) fit on the screen and b) show all the data. If both aren't possible (ie small screen or odd shape) then the pop-up should be scrollable and should fit to screen size (or rather, browser area)

magicbug, 2019-10-08:
While it's not a bug and nor incorrectly formatted, the window is too small and that needs to be adjusted, I suspect that will be by making it nearly full height.

magicbug, 2019-10-21:
On considering this issue, I'm purposing that instead of the QSO Edit popping up in a window that its actually a standard page, this is because the amount of data needing to be shown is starting to outstrip space!

motoionic, 2019-12-25:
Agree the edit window is a little funky at times and a full page would be perfect.  I've found on mine, when I go back in to add extra details such as the SOTA reference or fix signal reports, etc. that it create a new QSO record instead of 'editing' the existing one, so I end up with a duplicate and have to then go in and delete the original.

closed 2020-03-01 by magicbug