magicbug/Cloudlog #377: eQSL imoprt/export broken?

g0wfv, 2019-10-18:
**Describe the bug**
Importing from eQSL gives no indication of success or failure (blank screen)
Exporting finds no QSOs to send (despite QSOs waiting to be sent!)

**To Reproduce**
Steps to reproduce the behavior:
Try to use any of the functionality in any way, and it fails

**Expected behavior**
eQSLs to be downloaded and uploaded

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Ubuntu 19.04
 - Browser: Chrome
 - Version: 
Google Chrome | 77.0.3865.120 (Official Build) (64-bit)
Revision | 416d6d8013e9adb6dd33b0c12e7614ff403d1a94-refs/branch-heads/3865@{#884}
OS | Linux
JavaScript | V8 7.7.299.13

magicbug, 2019-10-18:
Has QTH Nickname for the station profile been provided?

g0wfv, 2019-10-18:
Yes.

Export seems to be workig, ot sure what was going on there.
Import doesn't ....

LOTW import seems to have broken too .....

magicbug, 2019-10-18:
OK thats weird, I'm testing with my logs and it seems to be functioning and I know @dg9vh looked at the code the other week.

Guess need to debug importing!

magicbug, 2019-10-18:
Stupid question but you can write in ./uploads/?

magicbug, 2019-10-18:
My guess would be that your username and passwords wrong.

I've added some crude debugging, when it white screens it will give you the url its using to download from, try in the browser, eQSL might return an error

g0wfv, 2019-10-18:
Looks like the eqsl qth nickname isn't being passed in the query string ....

http://www.eqsl.cc/qslcard/DownloadInBox.cfm?UserName=g0wfv&Password=XXXXXXXX&RcvdSince=1970/01/01&ConfirmedOnly=1

magicbug, 2019-10-18:
Isn't needed for eQSL download, I just take the lot

g0wfv, 2019-10-18:
Seems it is -

Changed that link so it has my password in it ... FAILED

Added the QTHNickname to the querystring and it worked

Error: Username/Password found more than 1 accountPlease specify the
desired User by using the QTHNickname parameter

magicbug, 2019-10-18:
oooh

magicbug, 2019-10-18:
That be why then, seems you can't take a full download for ALL if more accounts are linked..

OK need to adjust that :)

g0wfv, 2019-10-18:
Yeah I'm not a massive fan of eQSL's QTH Nickname thing - I tried it years ago, didn't get on with it, but now can't get rid of it!  All I can do is have a QTHNickname of "All" now :-/

magicbug, 2019-10-18:
Just writing code to grab it from the active station profile, fix in 10mins

g0wfv, 2019-10-18:
lol I was about to offer to do it, as it's the same issue I fixed with my first code offering to the project ;-)

magicbug, 2019-10-18:
@g0wfv you can debug it and fix it if i screw it up :p

g0wfv, 2019-10-18:
lol will do ..... ;-)

Seen the commit in git already :-)

magicbug, 2019-10-18:
OK, see latest commit.. shout if it works lol

g0wfv, 2019-10-18:
A PHP Error was encountered
Severity: Notice

Message: Undefined variable: station_profile

Filename: controllers/Eqsl.php

Line Number: 134

An uncaught Exception was encountered
Type: Error

Message: Call to a member function row() on null

Filename: /var/www/html/cloudlog/application/controllers/Eqsl.php

Line Number: 134

magicbug, 2019-10-18:
try now, think I missed something

g0wfv, 2019-10-18:
An uncaught Exception was encountered
Type: Error

Message: Cannot use object of type stdClass as array

Filename: /var/www/html/cloudlog/application/controllers/Eqsl.php

Line Number: 146

magicbug, 2019-10-18:
getting closer :p

g0wfv, 2019-10-18:
That one got it ....

g0wfv, 2019-10-18:
All seems to be working again now :-) Thanks :-)

closed 2019-10-18 by g0wfv