magicbug/Cloudlog #378: ADIF import needs more data validation/cleanup

Maescool, 2019-10-21:
When testing the import feature, I got an error on COL_LAT or COL_LON (forgot exact which one)
That it went out of bounds.
Checked, seemed some float value ending 0000001, which made it more than varchar(15)

So, maybe adding some more sanity checks/cleanup to the fields would be nice.
Never trust user input, that also counts for ADIF files.

magicbug, 2019-10-21:
ADIF can't ever be trusted but predicting what logs going to provide is even harder LAT long per spec

> sequence of 11 characters representing a latitude or longitude in XDDD MM.MMM format

The question then is should it not import the QSO if it's not valid. chew off the extra length and still import.

magicbug, 2019-10-21:
But yes will add in a santity check for this field to make sure that it strips anything beyond the 11 charaters limit using substr() function in php

magicbug, 2019-10-21:
Just added limiting for Lat, Lon, My_Lat, My_Lon

Maescool, 2019-10-21:
I would say, try to at least parse it with the known ways people tend to write down those fields (yeah, that is a total pain)
The minimal I would like is, not getting database errors, so checking it is not overflowing fields and is the right datatype, is a minimum I would guess

magicbug, 2019-10-21:
Just to further update `a_index, age, ant_az, ant_el, ant_path` have had similar validation, I'll slowly work my way down the list :)

closed 2019-10-28 by magicbug