magicbug/Cloudlog #379: Creation station profile failed ...

lotrdocker, 2019-10-22:
Hi,

I try to create a station profile but the creation failed with the error "**Incorrect integer value: for column 'station_dxcc' at row 1**". I can't specify anything in the form for the **Station DXCC** section.

Create station profile screenshot:

[image]

Thank's for your help.

magicbug, 2019-10-22:
Per installation guide was "update country files" within the admin drop down carried out?

lotrdocker, 2019-10-23:
Hi Peter,

My fault, I did'nt do "update country files" step. All is ok now. Thank you.

closed 2019-10-23 by lotrdocker

dg9vh, 2019-10-23:
Idea for future: why not fill the table initially on install with create script for db?

magicbug, 2019-10-23:
I actually used to, but it ended up users never updating it, so I feel
having them download on first install isn't much to ask