magicbug/Cloudlog #380: Change Country Names on Dashboard/Logbook Display to Flags

magicbug, 2019-10-22:
I'm proposing that on the Dashboard/Logbook page where countries are referenced that they are replaced with suitable flags to improve space and layout available.

The country name would still be shown when hovering over said flag.

tonyc, 2019-10-22:
I think the issue with this would be that you instantly lose readability, and at least in the US, I'd suspect that most people don't know what other country's flags look like, outside of a handful. This is probably less of a problem in EU, perhaps?

I'm scanning down the flags on https://en.wikipedia.org/wiki/List_of_ISO_3166_country_codes and I would be terribly confused, so I'm wondering if there's a better way besides a flag.

Could you attach a screenshot of what it looks like currently?

magicbug, 2019-10-22:
Interesting, on this page, I don't think it would make a huge difference unless most don't know where a callsign references? You'd still get full dxcc name when making the QSO.

I could make it an option, you can have full name or the flags? half way keep everyone happy. maybe EU is more aware of callsign and country.

[image]

tonyc, 2019-10-22:
Honestly, on that screenshot, I think it looks fine the way it is.

tonyc, 2019-10-22:
Ah yes. Hmmm...... what about collapsing it to a flag on mobile devices, and use the full country name if there's enough width?

magicbug, 2019-10-22:
I think that seems like a reasonable halfway medium, as always easy to think theres an issue when there isn't!

closed 2019-11-25 by magicbug