magicbug/Cloudlog #381: Feature: Bandmap

magicbug, 2019-10-22:
I think the FR of a band map probably appears at least weekly in my inbox, most requests are for DX Cluster and RBN.

The actual timeline side isn't that hard with CSS, the more technical issue is getting the spots to it, well actually that's not hard with web sockets, RBN poses a sheer number of spots issue.

What's everyone's thoughts, ideas on its implementation?

AndreasK79, 2020-10-29:
Are you also thinking that the bandmap should show needed status on each spot? If so, wouldn't this generate several queries to the database for each spot?

magicbug, 2020-10-29:
Yes would need to show cluster spots, my ultimate plan is to use web sockets for this so I have a "service" chucking out the spots and then javascript handles the rest so zero real database apart from perhaps a check to see if its needed if its a "new spot"

AndreasK79, 2020-10-30:
I understand. Seems like you have things figured out here. I see this as a useful feature, but I don't think I can pull of this one.

daccle, 2020-11-09:
Are there any parts in the application which already uses web sockets? Besides I have no clear understanding how this could/should look like. I willing to help on this, but at the given time, I am not feeling confident enough on the code base to do this on my own.

closed 2020-12-11 by magicbug