magicbug/Cloudlog #382: Create database problem.

EA8CXN, 2019-10-25:
Hi, first of all I have to say that I'm a newby , and I have limited knowledge on programing.
I already install the CloudLog on a raspberry pi , just only for testing purpose and everything work well.
Now, next step is to install into my host, but, there is a problem:
- My hosting do not allow to create a database from a php code.
I can create and empty database, and also create ussers. My question is , Can I change the code to look directy to a database already created by me?

Thanks

magicbug, 2019-10-25:
The Cloudlog installer doesn't create a database, you supply the database name and logins of an empty DB and then it executes SQL to create tables, I'd be surprised if any host didn't allow that?

Peter.

EA8CXN, 2019-10-26:
Ok, understand little bit more.
By the moment the only thing that I can do is to import the .sql file on assets with PHPmyADMIN, so the tables are already created but I don't know if is enough with this, or I have to modify several files by hand. I'm checking all files in CloudLog and change manually to my database settings (DSN, database name...) But no good results... Maybe I broke the code so I have to start again.

magicbug, 2019-10-26:
If you imported the SQL found in the install/assets folder you would then need to setup the two configuration files.

I'd highly recommend using the install system, its reliable & works on all shared hosting environment especially if you're not experienced with PHP/mysql

EA8CXN, 2019-10-26:
Ok, finally I resolve the error.
My problem is that in my host, there is a port and I always insert it on the installation system.for example: xxx-dns.net:3507 ... So after removing the port, everything is good.

Now another problem appears, when I try to create the "Station Profile" I get the next message:
**_Error Number: 1366

Incorrect integer value: '' for column 'station_dxcc' at row 1

INSERT INTO `station_profile` (`station_profile_name`, `station_gridsquare`, `station_city`, `station_iota`, `station_sota`, `station_callsign`, `station_dxcc`, `station_country`, `station_cnty`, `station_cq`, `station_itu`) VALUES ('QTH', 'IL18SK', 'La Matanza de Acentejo', 'AF-004', '', 'EA8CXN', '', '', 'SPAIN', '33', '36')

Filename: models/Stations.php

Line Number: 49_**

EA8CXN, 2019-10-26:
Ok, everything solved now.
The previous error was because:
- Permission folder issues
Thanks

closed 2019-10-26 by EA8CXN