magicbug/Cloudlog #384: Distance Chart stats page

M0TZO, 2019-11-02:
This is an idea for a chart to show distances worked.

Similar to the grid square page, maybe using grid squares to work out distances appropriately, per band, mode & overall.

AndreasK79, 2020-02-10:
Are you thinking of something like this? [https://la8aja.com/distplot/](https://la8aja.com/distplot/)
This is something I wrote a while back. It uses the gridsquare for calculation. I might be able to implement something like this in Cloudlog.

[image]

magicbug, 2020-02-10:
Even if thats not what @M0TZO had in mind, i wouldn't mind that feature for searching my satellite QSOs to see distances :)

AndreasK79, 2020-02-11:
@magicbug I'll see what I can do. It's mostly a matter of adapting the code to work with MVC in Cloudlog, so it's not a total rewrite.

AndreasK79, 2020-02-12:
@magicbug I have a working concept:
[image]

I need to polish it up a little bit, but right now I have the code fetching all the bands in config. How do you select on satellite QSOs? Should I also assume that col_my_gridsquare is always set (just validate the gridsquare, as I do for gridsquares fetched from each qso)?

magicbug, 2020-02-12:
prop_mode column when set to "SAT" will provide all satellite QSOs.

I'd use the grid square for the active profile for the "operators square" rather than any field from the qso table.

Within QSO there's two grid square fields ones `COL_GRIDSQUARE` which has just one square but if the QSO is for a grid line more than one square then it's stored in `COL_VUCC_GRIDS `this is the same for even VHF/UHF QSOs those in `COL_VUCC_GRIDS` if you look in `application/models/Gridsquares_model.php` you will find some example database stuff that I use for building the gridsquare maps.

magicbug, 2020-02-12:
Just a follow on satellite name assigned to a qso is in COL_SAT_NAME

commit 3e9132e1 (2020-02-20): Merge pull request #418 from AndreasK79/Distanceplotting

Added distanceplotting. Fixes #384.
commit e6c788a1 (2020-02-20): Added distanceplotting. Fixes #384.
closed 2020-02-20 by magicbug