magicbug/Cloudlog #385: password not accepted - is all lost ???

DL3RTL, 2019-11-09:
I use couldlog since 3 months and changes nothing. all of a sudden I can not log in to logbook from any device. how can I reset the password? how can I get all my data??? need HELP !!!

magicbug, 2019-11-09:
Fear not, go to /application/config/config.php then find  $config['use_auth'] = true; change to false reset password & set it back to true.

DL3RTL, 2019-11-09:
Many thank for your fast reply Peter. I did change it to false and stored the file. Then i changed it again to true and stored the file once again on the server.
What do I have to do next  to log in or to reset/change the password ?

dg9vh, 2019-11-09:
Hi,
I think the process would be:
- set it to false
- login to cloudlog and go to "edit user" to change your password
- set it to true again

73 de Kim
DG9VH

DL3RTL, 2019-11-09:
But I can not log in without the password.

DL3RTL, 2019-11-09:
In the URL https://www.   ........   /index.php/user/login

I type my Username and leave the password field blank.

Then I get the error message "The Password field is required.

Oh my god.....

dg9vh, 2019-11-09:
Are you able to access to the database directly via phpmyadmin or something like that?

DL3RTL, 2019-11-09:
yes

dg9vh, 2019-11-09:
So you can use https://php-password-hash-online-tool.herokuapp.com/password_hash for example to generate a initial password like test01 and paste the password-hash into the users-table-row concerning your user. After this, you could try to log in with test01 as password.

dg9vh, 2019-11-09:
you could use following parameters for generating - I tested it on my dev-system, it's working fine:
[image]

magicbug, 2019-11-09:
@dg9vh's method works, however setting Auth to false should make huge admin drop down available then just edit

DL3RTL, 2019-11-09:
I tried it with test01 but I can not log in :-(

DL3RTL, 2019-11-09:
And what is a huge admin drop down ?

DL3RTL, 2019-11-09:
Now it worked with the new password. I forgot to set to true again in the config file. I sucessfully changed my password again and I have full access to the logbook.

Still I don't know what went wrong. I was just storing a qso when I was logged out and had no chance to get in again.

Never the less many thanks for your fast help that late evening Peter and Kim.

Daniel...

dg9vh, 2019-11-09:
Ok, then I can go to bed in a good mood :-)

Have a good night :-) Best 73
Kim

closed 2019-11-11 by magicbug