magicbug/Cloudlog #387: QSO Time Incorrect

tspowell01, 2019-11-12:
I just installed Cloudlog on my Linux box.  I clicked on the QSO and clicked on Live QSO.  When the screen came up, I noticed that the time was off by 5 hours; I'm on eastern time in the USA.  I edited the profile for the account I was using, but changing the time zone didn't change the time on the Live QSO screen. My Linux box time is set to local time.  Any suggestions? Thank you.

magicbug, 2019-11-13:
From memory Cloudlog should default to UTC, this is the first instance that that not happening that I'm aware of.

Just to confirm that the date/time on the servers correct? Could you screenshot the QSO panel too.

The QSO info in the browser is automatically figured out based on local clock of the browsers device it doesn't actually use the time zone selection.

Info on timezone used also be useful & browser.

magicbug, 2019-11-13:
Quick further comment, I'm going to guess you haven't changed the time zone info in application/config/config.php

tspowell01, 2019-11-13:
Okay, it's a misunderstanding on my part.  I was expecting the time in the QSO to be the same as my local time, but you convert it to UTC.  

[image]
[image]

closed 2019-11-13 by tspowell01