magicbug/Cloudlog #388: Database error creating new user profile

ClaudeMa, 2019-11-13:
Database error creating new user profile
Input field station_dxcc not exist
Upgraded for previous version of cloudlog

> A Database Error Occurred
> 
> Error Number: 1366
> 
> Incorrect integer value: '' for column `f4ikh`.`station_profile`.`station_dxcc` at row 1
> 
> INSERT INTO `station_profile` (`station_profile_name`, `station_gridsquare`, `station_city`, `station_iota`, `station_sota`, `station_callsign`, `station_dxcc`, `station_country`, `station_cnty`, `station_cq`, `station_itu`) VALUES ('Mycall', 'JN15SU', 'MyCity', '', '0', 'Mycall', '', '', 'MyContry', '14', '27')
> 
> Filename: models/Stations.php
> 
> Line Number: 49

All field are provide but there is no input field for DXCC

 - OS: LWindows 10 x64
 - Browser Firefox
 - Version70

Server
Debian Buster x64
MariaDb 10
PHP 7.3

[image]

magicbug, 2019-11-13:
As per the wiki instructions did you go to admin -> update country files this needs to be done before creating station profiles

ClaudeMa, 2019-11-15:
Thx work now

very good job :-)

closed 2019-11-15 by ClaudeMa