magicbug/Cloudlog #392: update issue

g7kse, 2019-11-30:
I'll start this by saying it was probably an error on my part. Updated on self hosted cloudlog and get most of the menu's like this

[image]

with the exception of when I go to admin and it looks all swizy, new and lovely, a la....

[image]

Any chance of putting me right? It seems perfectly functional though.

I also noticed that you've got a link to when I poped it onto a Pi Zero W. I'm going to update this as I'm in the process of moving everything onto my own server so the link will not work for now. I've moved everything onto a DietPi distro which includes a web server and it is a much nicer option, just don't expect a brilliant Boostrap based site. it is me after all ;-)

magicbug, 2019-12-01:
Top versions old and the new ones how it looks now. If you'd like to add anything to the wiki feel free.

g7kse, 2019-12-01:
Sorted the wiki. Now killed my install.....A little knowledge is dangerous ;-)

closed 2019-12-01 by g7kse