magicbug/Cloudlog #393: Cloudlog not updating Country Files

Semir8, 2019-12-04:
Since I've installed Cloudlog in my Raspberry Pi, I'm not able to update the country files, as it gets stuck in Updating... but you could wait for hours, nothing will happen. I've set the permissions for uploads folder to 777 to try, but nothing changed. Any idea?

dg9vh, 2019-12-04:
[image]

Here working without problems.

magicbug, 2019-12-04:
I'd make sure that you have all the PHP modules installed as per the wiki documentation as this is likely server related

TheMcBain92, 2019-12-04:
Check your php permissions in .htaccess 

had a similar issue to this. It is on of the issues listed on here somewhere.

abrain, 2019-12-04:
> had a similar issue to this. It is on of the issues listed on here somewhere.

Here it is: https://github.com/magicbug/Cloudlog/issues/357

ea1jay, 2019-12-04:
All PHP modules listed on the wiki are installed, and I'm running nginx (it's a self-hosted server on a Raspberry Pi) so, if I'm not wrong, no .htaccess.

The site configuration file looks like this:

server {
	listen 80 default_server;
	listen [::]:80 default_server;

	# SSL configuration
	#
	# listen 443 ssl default_server;
	# listen [::]:443 ssl default_server;
	#
	# Note: You should disable gzip for SSL traffic.
	# See: https://bugs.debian.org/773332
	#
	# Read up on ssl_ciphers to ensure a secure configuration.
	# See: https://bugs.debian.org/765782
	#
	# Self signed certs generated by the ssl-cert package
	# Don't use them in a production server!
	#
	# include snippets/snakeoil.conf;

	root /var/www/html;

	# Add index.php to the list if you are using PHP
	index index.php index.html index.htm index.nginx-debian.html;

	server_name _;

	location / {
		# First attempt to serve request as file, then
		# as directory, then fall back to displaying a 404.
		try_files $uri $uri/ /index.php?$args;
	}

	# pass PHP scripts to FastCGI server
	#
	location ~ \.php$ {
		include snippets/fastcgi-php.conf;
	#
	#	# With php-fpm (or other unix sockets):
		fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
	#	# With php-cgi (or other tcp sockets):
	#	fastcgi_pass 127.0.0.1:9000;
	}

	# deny access to .htaccess files, if Apache's document root
	# concurs with nginx's one
	#
	#location ~ /\.ht {
	#	deny all;
	#}
}

ea1jay, 2019-12-05:
Tried to copy php.ini from /etc/php/7.0/fpm/ to the updates and uploads folder. Same thing happens, no update. 
[image]

magicbug, 2019-12-05:
All I can really suggest is checking server logs as this isn't a Cloudlog related issue.

ea1jay, 2019-12-05:
I've got this error in the log:

2019/12/05 19:25:27 [error] 664#664: *1548 upstream timed out (110: Connection timed out) while reading response header from upstream, client: MY_IP_ADDRESS, server: _, request: "GET /index.php/update/dxcc HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock", host: "cloudlog-address", referrer: "http://cloudlog-address/index.php/update"

Can anyone help me please?

magicbug, 2019-12-05:
Does the machine have access to the outside internet? The country files are downloaded?

magicbug, 2019-12-05:
Timeout suggests that something is timing out before it's processed if it's managed to download so could be an issue between PHP & the web server

ea1jay, 2019-12-06:
Finally I could solve it. Make sure the config file of the web server in /etc/nginx/sites-available looks like this:

```
location ~ \.php$ {
		include snippets/fastcgi-php.conf;
	#
	#	# With php-fpm (or other unix sockets):
		fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
	#	# With php-cgi (or other tcp sockets):
	#	fastcgi_pass 127.0.0.1:9000;
		
		
                fastcgi_split_path_info ^(.+\.php)(/.+)$;
	        include fastcgi_params;
	        fastcgi_param SCRIPT_FILENAME $document_root/$fastcgi_script_name;

	}

```

closed 2019-12-06 by ea1jay