magicbug/Cloudlog #395: Incorrect integer value: '' for column 'station_dxcc' at row 1

M0LTE, 2019-12-08:
**Describe the bug**
When creating a station profile, I get the following:

```
A Database Error Occurred
Error Number: 1366

Incorrect integer value: '' for column 'station_dxcc' at row 1

INSERT INTO \`station_profile\` (\`station_profile_name\`, \`station_gridsquare\`, \`station_city\`, \`station_iota\`, \`station_sota\`, \`station_callsign\`, \`station_dxcc\`, \`station_country\`, \`station_cnty\`, \`station_cq\`, \`station_itu\`) VALUES ('Home IO91lk', 'IO91LK', 'Reading', '', '', 'M0LTE', '', '', '223', '14', '27')

Filename: models/Stations.php

Line Number: 49
```

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade cloudlog to a version that prompts you to create a station profile
2. Create a station profile with the above data

**Expected behavior**
A station profile to be created

I worked around it by manually creating a row in the database.

Cheers Peter.

magicbug, 2019-12-08:
That must have been a fairly old version as station profiles have been available since September, it looks like Country Name is whats missing, this is populated using the info from Club log via the country files, when you were creating the station profile did country name populate the dropdown?

M0LTE, 2019-12-08:
No, from memory it didn’t render an input field at all.

I just did a git pull to get up to date- it’s been a while since I’ve done
that.

dg9vh, 2019-12-08:
Sometimes it would help to clean cache in browser...

magicbug, 2019-12-08:
This is what the station profile (Add) page looks like, if it doesn't then make sure "Update Country Files" is carried out.

[image]

M0LTE, 2019-12-08:
> [image:
> screencapture-cloudlog-mg-index-php-station-create-2019-12-08-21_44_28]

It now looks like that after updating country files.

closed 2020-01-22 by magicbug