magicbug/Cloudlog #399: Keeping login details

motoionic, 2019-12-25:
My station especially today I've noticed that when I keep coming back to the browser, I'm logged back out of cloudlog.  Is there a way to keep the login session for longer or indefinitely ?  Seems to be losing the login after 15 or so minutes of inactivity on the logbook.

I've tried resetting Chrome details for the website incase it was deleting a cookie or something, but I'm not seeing anything of interest there.

magicbug, 2019-12-25:
Ultimately the login sessions are either driven by local cookies or the database depending on how you set things up, inside application/config/config.php you do have some options on session times.

closed 2019-12-26 by magicbug

motoionic, 2020-01-19:
I've been experimenting with the values in the config.php file, but I can't see to get anything to hold the login for any length of time.

Its so annoying when you have post QSO open ready for a SOTA activation in my local area, fill out the form only to then be logged out when you click Save QSO.

I'm wondering if my shared hosting and having the option set to files is causing an issue?  If I opted for database is there a table already created in the schematic.  I'm not familiar with codeignitor and how its running sessions.

I just need like a Remember Me option to keep me logged in for the day and not keep throwing me out every few minutes.

magicbug, 2020-01-22:
Sessions will last by default around 120-180 minutes, it will actually be more than that if you are logging qsos within that time period, you're actually the first time I've heard of someone being constantly logged out, which hints that this might be browser-related, so remember feature will have a similar issue.

motoionic, 2020-01-24:
I'm wondering if its something to do with the way PHP config is setup on the server.  I use several browsers, Android phone, Mac OS, Windows 10, Ubuntu at the radio club, Fire tablet. depending on where I am. Will have a play with the session settings on the php.ini as well.