magicbug/Cloudlog #403: Error when submitting QSO

ofadam, 2020-01-22:
I'm getting the following error when submitting a QSO:

A Database Error Occurred
Error Number: 1525

Incorrect DATE value: '0000-00-00'

SELECT `call`, `entity`, `adif`, `cqz` FROM `dxcc_prefixes` WHERE `call` = 'KN4YNK' AND (`start` <= '2020-01-20 00:37' OR start = '0000-00-00' OR start is null) AND (`end` >= '2020-01-20 00:37' OR end = '0000-00-00' OR end is null)

Filename: models/Logbook_model.php

Line Number: 1486

magicbug, 2020-01-22:
Have you gone to Admin - Update Country Files since installing? Just tried
the QSO here and don't get the same error.

Peter.

ofadam, 2020-01-23:
I've done that. I'm running MySQL 8.0 and it looks like later versions don't accept 0-dates for null by default. Might that be the cause of the error?

If so, a user would need to change the default MySQL settings or the software would need patched to use non-0 dates.

dkozinn, 2020-02-08:
Same issue here, though it happened on import (which I tried to do before logging a QSO). Also running MySQL 8.0. I've worked around this for now by removing it from sql_mode in mysql.cnf, but according to the documentation:

> Because NO_ZERO_DATE is deprecated, it will be removed in a future MySQL release as a separate mode name and its effect included in the effects of strict SQL mode.

This would more of an issue in the future as disabling strict SQL mode might be undesirable.

ofadam, 2020-02-10:
Submitted pull request that may solve it: https://github.com/magicbug/Cloudlog/pull/413

magicbug, 2020-02-12:
Merged hopefully solved

closed 2020-02-12 by magicbug