magicbug/Cloudlog #406: Display name , QTH, DOK from earlier QSO and fill in

do2ih, 2020-01-25:
First of all: many thanks for a great piece of software.

We like to use the program for logging the rapports after our weekly ham radio broadcast messages.
If I enter the callsign, the program should display (and fill in) the name, the QTH and maybe the DOK, if the callsign is already in the database. Being able to address the other OM is something very useful, not only for broadcast logging.

73, DC1KW (Ingmar)
ex DO2IH

magicbug, 2020-01-25:
This is already a core feature, past info is pulled from the logbook if it's a new callsign you can either look it up from qrz or hamqth.

closed 2020-01-25 by magicbug

do2ih, 2020-01-28:
After updating the country files it works as aspected.

Thank you
73, de DC1KW