magicbug/Cloudlog #407: Uploading lotw report; no qso found

m0pcb, 2020-02-03:
**Describe the bug**
When uploading a lotw.adi report from a file, if the QSO record doesn't match it appears that a new QSO is added in the active station profile with the QSO details from the lotw.adi file. 

**Expected behavior**
If the QSO in the lotw report.adi doesn't match anything in the log then there should be a warning and the data not inserted into the active profile.

**Desktop (please complete the following information):**
Hosted instance on cloudlog.co.uk

m0pcb, 2020-02-03:
Hmm, it actually seems to have reassigned the QSOs to the active profile. This happened to 17 QSOs in a large upload, where the QSOs were logged as M0PCB/P but lotw had them as M0PCB. The result was that the QSOs were then attributed to AD5XI profile.

magicbug, 2020-04-06:
Not sure why you'd upload the report as it would treat the QSO as new.

Best way to match is via the built in system.

magicbug, 2020-04-06:
Hmm looks that that might be broken in that case, I know the internal system for downloading with user & pass works will need to explore

AndreasK79, 2020-04-28:
What about adding an option to not import QSOs not found? That way people can decide when pulling data from LoTW.

magicbug, 2020-04-28:
Yes I much prefer that solution

magicbug, 2020-04-28:
My biggest concern is when adding a missing QSO on import is that it gets correctly assigned to the right station profile in Cloudlog

AndreasK79, 2020-04-29:
@magicbug Ok. What about an option, and then check if current station_id matches the callsign from LoTW? If not, another check could be done to check to find a station_id with that call. If not, skip the whole import thing.

magicbug, 2020-04-29:
That feels like a reasonable compromise.

magicbug, 2020-04-29:
While thinking about it, can we match based on the callsign and the gridsquare from LoTW that should give a perfect match every time

PR #484 (2020-05-01): Added option if you want to import missing QSOs. If yes, try to locat…
…e station_id. If not found, do not import. Chose to use only 4 chars in gridsquare, and like in query, to have the highest success in finding a station_id, since 6 char gridsquares can be entered, as well as multiple gridsquares. Fixes #407.
commit 98954942 (2020-05-01): Added option if you want to import missing QSOs. If yes, try to locate station_id, if missing, do not import. Chose to use only 4 chars in gridsquare and like in query, to have the highest success in finding a station_id, since 6 char gridsquares can be entered, as well as multiples gridsquares. Fixes #407.
closed 2020-05-01 by magicbug