magicbug/Cloudlog #409: Failure during installation due to missing mysqli class

dkozinn, 2020-02-07:
**Describe the bug**
Initial installation fails. An empty page is returned. An error is found in the Apache error log (more below).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to \<url-for-cloudlog\>/install
2. Fill out form
3. Click on Install
4. Blank page is installed, no tables are created

**Expected behavior**
Expected behavior is that Cloudlog will be initialized, tables created in the database, etc.

**Desktop (please complete the following information):**
 - OS: Ubuntu 19.10 (Server), Windows 10 (Client)
 - Browser: Chrome
 - Version: Cloudlog master branch installed via ```git clone git@github.com:magicbug/Cloudlog.git```

**Additional context**
This error is found in the Apache error log:
```[Fri Feb 07 01:04:56.420890 2020] [php7:error] [pid 23327] [client 192.168.0.3:26852] PHP Fatal error:  Uncaught Error: Class 'mysqli' not found in /var/www/html/cloudlog/install/includes/database_class.php:9\nStack trace:\n#0 /var/www/html/cloudlog/install/index.php(22): Database->create_database(Array)\n#1 {main}\n  thrown in /var/www/html/cloudlog/install/includes/database_class.php on line 9, referer: https://<myhost>/cloudlog/install/```

php -m shows mysqli is installed

I'm not sure what else to look at, any assistance would be appreciated.

73 de K2DBK

magicbug, 2020-02-07:
That error is saying that the modules not installed, do you have two versions of PHP installed? 

You could try <?php phpinfo(); ?> In a empty PHP folder and see what modules are listed there.

dkozinn, 2020-02-07:
I've got it installed under /var/www/html/cloudlog, there's no PHP in that path.  Running a script with phpinfo() shows that mysqli is available, and there only seems to be a single path for extensions. If it would help, I can privately send you the info to see the full output of them.

I'm also not quite sure what you mean by trying in an empty PHP folder. Apologies for being a bit dense, my PHP-foo is a bit rusty.

dkozinn, 2020-02-08:
Dumb user error: I'd installed everything but never restarted Apache. Oops. All better now.

closed 2020-02-08 by dkozinn