magicbug/Cloudlog #414: Changing "station" to "none" results in lost QSO

ofadam, 2020-02-10:
If accidentally changing an existing QSO's station from a defined one to "none," the QSO disappears completely from the Cloudlog web interface. That said, it still exists in the database under "TABLE_HRD_CONTACTS_V01."

Would there be a way to make sure that either entries with the "none" location marked are still visible via web interface? If not, could it be removed as an option?

magicbug, 2020-02-10:
OK thats a bug, None should default to whatever the active one is set to in Station Profile, the station profile in the QSO panel is from an early implementation but felt it might be an idea to keep it for quickly having a qso from a different location.

AndreasK79, 2020-02-12:
Why is none even an option then? Wouldn't it be better to just remove it in the select?

magicbug, 2020-02-12:
It was an option 5 years ago :) but yes just needs removed and to select the active profile

commit 0628e3bd (2020-02-17): Merge pull request #416 from AndreasK79/Removednonefromprofileselect

Removed the ability to set none in stationprofile. Fixes #414
commit a57edd03 (2020-02-17): Removed the ability to set none in stationprofile. Fixes #414
closed 2020-02-17 by magicbug