magicbug/Cloudlog #417: can no longer log in

nr0q, 2020-02-16:
After the nightly "auto-update", my install of Cloudlog no longer allows me to log into the program. I go to the page and can see my log but the user/password field and login button are all missing

Installed on my webserver running CentOS 7 and Apache with MariaDB

nr0q, 2020-02-16:
one thing to note is that GridTracker is still able to push data into my log through the API

dg9vh, 2020-02-16:
Can you show us some screenshots of the situation?

magicbug, 2020-02-16:
Nothings changed in the code to effect login, so something else has gone a miss.

You can disable the auth in the application/config/config.php set $config['use_auth'] = true; to false then reset the password and set it back to true.. however I'm interested to know what caused a lock out.

nr0q, 2020-02-16:
sure, here's screen shots of the web browser tab, database and a 'ls -l' of the main directory

[image]
[image]
[image]

nr0q, 2020-02-16:
I tried to change $config['use_auth'] = true; to false and restarted the httpd service just in case and still wouldn't let me in.

magicbug, 2020-02-16:
Sure your using the master branch of Cloudlog? That looks like v2 if that's the case it's not ready for use

nr0q, 2020-02-16:
should be the master, might have to re-clone it back and see if that fixes it

magicbug, 2020-02-16:
Quick give away would be doing /login and /register URLs and if it returns something that looks different then that will give it away

nr0q, 2020-02-16:
those urls returned 404 errors, I accidently blew away the Cloudlog folder, so no cloning it back but am getting a bunch of error messages, i think i'm going to have to wipe the database and start over from scratch :(

magicbug, 2020-02-16:
The hrs table is safe plus the profiles but definitely sounds like you switched branch no idea how if you was automating it

nr0q, 2020-02-16:
i followed the directions on the wiki, so I'm not sure, but i probably will omit that step once I get it working again

nr0q, 2020-02-16:
I can longer get it running, i've wiped the database and started a new one, wiped the directory and did a new git clone and can't get anything to work, it once worked on this system and now nothing

nr0q, 2020-02-16:
so the only way I could find to repair an install that accidently got upgraded to V2 is to wipe everything and reinstall, for the record. 

issue closed

closed 2020-02-16 by nr0q