magicbug/Cloudlog #419: Cloudlog CAT IC7300 not reading frequency

g7kse, 2020-02-22:
Afternoon, I've been wrestling with this for a bit and haven't found a solution, only a partial one. My IC7300 is my main HF rig and is connected via a USB to the shack PC. It is notorisourly painful for it to work as a straightforward CAT interface regularly. It seems to have a mind of its own and a myriad of settings that don't offer obvious answers.

I don't think CloudlogCAT is the issue, nor OmniRig (as other users seem to have had some success with other software) but something that might need ironing out to save any future woes. Especially as it is a popular rig.

It will get this...

[image]

Using this setup 

[image]

So far I can get it to recognise the mode. Thats a new win, but the frequency read is missing. 

The usual drivers uninstall / reinstall and swapping over cables hasn't worked as has a lengthy trawl of the the usual help sources. 

If I find a solution I'll let you know and add it to the wiki. Not expecting any direct involvement in troubleshooting. Just highlighting an issue that could crop up again

Alex

magicbug, 2020-02-22:
Known issue with omnirig I need to read the Freq from vfo a I believe if I remember right

magicbug, 2020-02-22:
Temp fix is to modify the omnirig config, however currently in the process of rewriting Cloudlog CAT hopefully to dump omnirig requirements.

magicbug, 2020-02-22:
Closing this issue as it's not related to the web codebase but rather the desktop app please use the issues system on that repo.

closed 2020-02-22 by magicbug

g7kse, 2020-02-22:
Will do.