magicbug/Cloudlog #421: drop-down menus missing

0x6d61726b, 2020-02-26:
**Describe the bug**
After first installation and setup of cloudlog (latest master version), the drop-down menu, that shall be displayed after login, is missing. I however can setup a new user by manually opening `/index.php/user`.

**To Reproduce**
Steps to reproduce the behavior:
1. Extract https://github.com/magicbug/Cloudlog/archive/master.zip
1. Setup nginx site
1. Create database user and database
1. run `/install/`
1. update `config.php` (... `$config['base_url'] = 'http://<my-fqdn>/';` ...)
1. login using `m0abc` and `demo`

**Expected behavior**
Drop-down menu and login information is expected as shown [here](https://i1.wp.com/www.magicbug.co.uk/wp-content/uploads/2019/07/cloudlog-dashboard.png?resize=1080%2C928&ssl=1).

**Screenshots**
[image]
[image]

**Server:**
 - OS: Debian 10 buster (in virtual machine)
 - web server: nginx/1.14.2
 - PHP: 7.3.14-1~deb10u1

**Desktop (please complete the following information):**
 - OS: Windows 10 (64 bit)
 - Browser: Firefox 73.0.1 (64 bit), same for Microsoft Edge

Can somebody tell me what I did wrong?

dg9vh, 2020-02-26:
Hi, did you create at least one station profile in the administration tab?

0x6d61726b, 2020-02-26:
Hi Kim, I did not yet create a station profile because I cannot create a station profile (looks like #379).
It seems I have to [Update Country Files](https://github.com/magicbug/Cloudlog/wiki/Update-Country-Files) first, but don't know how to call it with the missing links. Can you tell me the link?

```
A Database Error Occurred
Error Number: 1366
Incorrect integer value: '' for column `cloudlog`.`station_profile`.`station_dxcc` at row 1

INSERT INTO `station_profile` (`station_profile_name`, `station_gridsquare`, `station_city`, `station_iota`, `station_sota`, `station_callsign`, `station_dxcc`, `station_country`, `station_cnty`, `station_cq`, `station_itu`) VALUES ('Home', '<MYLOC>', '<MYCITY>', '', '', '<MYCALL>', '', '', '', '1', '1')

Filename: models/Stations.php
Line Number: 49
```

dg9vh, 2020-02-26:
Hi, use this:
[image]

0x6d61726b, 2020-02-26:
I would expected the link being where you showed me in your screenshot, but unfortunately I have no drop-down menu at all (after login), neither is my user shown:
[image]

dg9vh, 2020-02-26:
maybe @magicbug has an idea...

magicbug, 2020-02-26:
Looks like you aren't logged in if you aren't getting the other menu items

0x6d61726b, 2020-02-26:
I get the `Search Callsign` entry after the login and I can create an user, so I would expected I am somehow logged in "partially"?

0x6d61726b, 2020-02-26:
*grr* found the issue. `$config['use_auth'] = true;` was set to false, because of "False turns all authentication off, best used when setting up".

Sorry for that stupidity of mine.

closed 2020-02-26 by 0x6d61726b