magicbug/Cloudlog #423: Leaflet markers based on mode.

mbridak, 2020-02-29:
I'd like to see different markers on the map based on the mode. 

Most of the work can be done in leafembed.js
example code snippet below.

var icn = plotlist[i]["html"].split(" ")[8];
var plotll = new L.LatLng(plotlist[i].lat,plotlist[i].lng, true);
if (icn == 'CW'){
	var plotmark = new L.Marker(plotll, {icon: cwicon});
} else if (icn == 'SSB'){
	var plotmark = new L.Marker(plotll, {icon: ssbicon}); 
} else {
	var plotmark = new L.Marker(plotll);
}
[image]

magicbug, 2020-03-01:
Code theory isn't the problem the issue is time to generate multiple icons for each mode in the hobby, there's actually already an issue for this already #202 which goes with the idea of using different colours which I think might be the better way unless someone wants to create icons for the ADIF mode list?

magicbug, 2020-03-01:
That said not ruling anything out, open to ideas and suggestions, if anyone has them!

nr0q, 2020-03-01:
How about different icons for modes and color them based on band. I'm
putting my thinking cap on right now trying to imagine what icons would go
for each mode. I can picture what CW, FSK and PSK might look like.

CW have a dit and dah inside the pinhead, FSK could be a pair of
alternating vertical lines and PSK the hourglass looking shape for when the
phase flips in the signal. I"ll get to playing with my graphics program
here after I finish some coffee.

Matthew Chambers, CBT, NR0Q
Owner/Engineer
*M Chambers Communications Engineering LLC*
PO BOX 855, Moberly, MO 65270
Mobile (660)415-5620
www.mchambersradio.com

nr0q, 2020-03-12:
Ok, so these aren't the prettiest icons I've ever made, but I'm thinking they wouldn't be up for interpretation and easy to read on a small screen. I made a sampling and attached them here. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

nr0q, 2020-03-12:
I also want to note, I'm working on creating the various colored ones, but am wondering how many different colors would we want/need? 

- LF/MF (3 bands)
- HF (9 bands)
- VHF (3-4 bands)
- UHF (3 bands)

This would depend on how complex or simple we want to code this. If i were coding it, I'd go for simple at least to start with. have dark red for MF and a lighter red for 160m, then increasingly lighter shades of orange through yellow for the HF bands, a light green for 6m, and darker for 4m, and even dark for 2m, then light blue for 70cm, and get darker as we go up in freq with purple for 23cm and up? But that's just me. I'd like some input from others including the person that posted issue #202 .

abrain, 2020-03-12:
> I also want to note, I'm working on creating the various colored ones

Creating a separate set of files per color is probably not necessary. If the icons are provided as SVG, they can be coloured through CSS. That also enables each instance to adapt the colors to their own needs.

But then there's still the question which colors to use initially. The following color scheme is used on https://pskreporter.info/, but I struggle to see a pattern behind it. Yours seems much more straight-forward.

[image]

mbridak, 2020-03-13:
I created a pull request with some code and icons. It's my first pull request so I THINK I did it right.

closed 2020-12-11 by magicbug