magicbug/Cloudlog #424: selected General -> Propagation mode setting doesn't stay for next QSO

AD7Djoe, 2020-03-02:
**Describe the bug**
For every  QSO I make, I still have to go to General -> Propagation mode -> Satellite.

I have QO-100 and SSB selected and these are sticky for each QSO.

If I do not set the propagation mode, then export, uploading to LOTW flags the entries as not having SAT mode selected.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

magicbug, 2020-03-02:
Is this in Live QSO mode or Post?

AD7Djoe, 2020-03-02:
Live QSO mode.

magicbug, 2020-03-02:
Interesting, I know for me when I populate Sat Name/Mode Cloudlog in the background will add Prop Mode, but i dont think it populates the frontend select but its certainly saved against the QSO as thats how I do it for QO-100

AD7Djoe, 2020-03-02:
OK… yea, thought it was all fine, but then LOTW flag me on some QSOs and sure enough, SAT wasn’t in those QSO so I went and edited the QSO, download, and upload to LOTW then they made it in.

magicbug, 2020-03-02:
While I couldn't see a bug with SAT_NAME not making prop_mode set, I have added code so that a session is set and Prop_mode is shown to track the previous QSO till cleared unless radio "satpc32" is selected then when you unselect the radio the prop mode is reset.

closed 2020-03-02 by magicbug