magicbug/Cloudlog #428: PHP Error

oe3ide, 2020-03-04:
Hi,
I tried to run cloudlog on my hosting-package (webspace, mysql databae). Install was successful, but I get some errors:

Severity: Warning
Message:  mysqli::real_connect(): Persistent connections are disabled. Downgrading to normal
Filename: mysqli/mysqli_driver.php
Line Number: 201

Severity: Warning
Message:  ini_set(): open_basedir restriction in effect. File(/tmp) is not within the allowed path(s): (/home/.sites/805/site1069704/web:/home/.sites/805/site1069704/tmp:/usr/share/pear:/usr/bin/php_safemode)
Filename: drivers/Session_files_driver.php
Line Number: 108

Any ideas?
73 de Ernst, OE3IDE

oe3ide, 2020-03-04:
Ok, first error (persistent connections) solved, I changed pconnect from true to false.

magicbug, 2020-03-05:
Session info can be changed in /application/config/config.php but this is normally fine. Sadly it's not a Cloudlog issue but a web host problem so you'll need to speak to them.

closed 2020-03-05 by magicbug