magicbug/Cloudlog #432: Downloading log form LoTW produces 503 Error

nr0q, 2020-03-11:
When I try to download an update for my log from LoTW I get
Message: file_get_contents(https://p1k.arrl.org/lotwuser/lotwreport.adi?login=user&password=password&qso_query=1&qso_qsl='yes'&qso_qsldetail='yes'&qso_mydetail='yes'&qso_qslsince=2020-03-11&qso_owncall=CALL): failed to open stream: HTTP request failed! HTTP/1.1 503 Service Temporarily Unavailable

Filename: controllers/Lotw.php

Line Number: 149
Error: Adif_Parser Already Initialized or No in ADIF File 

I went digging in the Lotw.php file but the first part of the URL "https://p1k.arrl.org/lotwuser/" is no where to be found. I need to change it to "https://lotw.arrl.org/lotwuser/" as they have shut off the p1k subdomain.

magicbug, 2020-03-12:
The URLs to my knowledge are stored in the database, I'm not aware of any reports saying the features not working. 

I'll double check the code but slightly confused.

magicbug, 2020-03-12:
Looks like URLs in the database will need changed to 'https://lotw.arrl.org/lotwuser' as the p1k are now unreliable and not always redirecting. changed in the schema will try to do migration file in a bit

Else login the database and change the items in the config table

magicbug, 2020-03-12:
Now fixed within the master branch, migration files been added to change the URLs.

closed 2020-03-12 by magicbug