magicbug/Cloudlog #433: Failed to boot 'VFS unable to mount root fs..' on Pi image version

g0cer, 2020-03-12:
**Describe the bug**
I downloaded the Pi distro which includes Cloudlog pre-installed from this site. 
On booting DMEG gets as far as :
[   1.54273 ---[ end Kernel panic - not syncing: VFS unable to mount root fs on unknown-block(179,2) ]---

Note I've transcribed that from the pi boot screen not pasted it but its accurate.
I have downloaded it twice and tried two seperate writer programs to write to the card in the Pi
Same error each time & multiple attempts at booting.

The Pi is Pi 3 Model  v1.2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
full boot

**Screenshots**
See attached photo of screen
[image]
.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
The Pi only has a wifi card in to connect to a keyboard - I have tried booting it without any
The Pi connects directly into a monitor via a HDMI >DVI interface which works normally this and other version.
The pi works normally with other OS cards in place.

magicbug, 2020-03-13:
Pi image is broken hence removed it from the wiki front-page, will work on improving it in the coming days.

closed 2020-03-13 by magicbug

g0cer, 2020-04-24:
Hi Peter 
I see the page is still on the wiki linked to the february version that won't boot as per above  
https://github.com/magicbug/Cloudlog/wiki/Raspberry-Pi-Image-Information
I must be the only one trying this or having this issue, I'd be interested to see if others have booted it.
Dave G0CER