magicbug/Cloudlog #437: clublog failed

dwhacks, 2020-03-15:
**Describe the bug**
Following the steps in the wiki for clublog just gives an error

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://<cloudlogserveraddress>/index.php/clublog/upload/<login>'
2. See "error"

**Expected behavior**
It to say "success"

**Desktop (please complete the following information):**
 - OS:Linux Mint
 - Browser Firefox
 - Version 73.0.1 (64-bit)

**Additional context**
Finally got my "logs" working and the log file says:
"ERROR - 2020-03-15 18:35:29 --> Clublog upload for VE7DNT failed reason" 
loglevel 4 does not divulge any more useful data.
the /uploads folder only has an index.html file in it

downloading from LoTW also doesn't work:
"failed to open stream: HTTP request failed! HTTP/1.1 400 Bad Request"

I'm very new to ham radio and logging apps, but have been running linux webservers for many years.

magicbug, 2020-03-15:
Have you confirmed the login information is correct? If so what environment is it hosted on PHP version numbers, have you installed all the modules as per the wiki.

nr0q, 2020-03-15:
Both LoTW and Clublog have had issues with being overloaded lately too. I'd
wait and try again later and see if they work. Also LoTW has very strict
rate limiting turned of, if you've recently visited the website on your
browser, the download in clublog will fail if you don't wait long enough.
How long we don't know as that's not been published.

Matthew Chambers, CBT, NR0Q
Owner/Engineer
*M Chambers Communications Engineering LLC*
PO BOX 855, Moberly, MO 65270
Mobile (660)415-5620
www.mchambersradio.com

dwhacks, 2020-03-15:
Running in Ubuntu server 18.04 LTS, fresh install on virtual machine, only running cloudlog.

php -v gives me 'PHP 7.2.24-0ubuntu0.18.04.3 (cli) (built: Feb 11 2020 15:55:52) ( NTS )'

also:
`php7.2 is already the newest version (7.2.24-0ubuntu0.18.04.3).
php7.2-curl is already the newest version (7.2.24-0ubuntu0.18.04.3).
php7.2-mysql is already the newest version (7.2.24-0ubuntu0.18.04.3).
php7.2-xml is already the newest version (7.2.24-0ubuntu0.18.04.3).
php7.2-mbstring is already the newest version (7.2.24-0ubuntu0.18.04.3).
`
I believe the login credentials are correct. I have tried my using my e-mail or my callsign in the profile section of cloudlog. I am able to log into club log with the same credentials, and have even tried their  "app passwords" option with no change. 

Everything else seems to work, including getting data from qrz (non paid) and updating country files

magicbug, 2020-03-15:
@nr0q I've seen no evidence of LoTW rate limiting in the last 8 years so interested to see any evidence.

As for clublog it will tell you on API call response whether that's the case and again not seen any cases of it.

magicbug, 2020-03-15:
@dwhacks thanks for the info, will have a think to what could be going on, I'm not seeing the issue here so makes it hard to hard to diagnose.

nr0q, 2020-03-15:
This was posted earlier this week on the ARRL-LOTW list
[image: lotw_rate_limited.png]
Matthew Chambers, CBT, NR0Q
Owner/Engineer
*M Chambers Communications Engineering LLC*
PO BOX 855, Moberly, MO 65270
Mobile (660)415-5620
www.mchambersradio.com

magicbug, 2020-03-15:
Ok so there was some temporary rate limiting that seemed to only effect a couple of days, most people self hosting won't hit the limit within Cloudlog sure to it not constantly polling LoTW.

dwhacks, 2020-03-16:
Any things that I can test for you? any ideas? Im a new club log user as well and only opened the account just yesterday.

Also, should I post info about the LoTW problem here or start a new?

magicbug, 2020-03-16:
@dwhacks setup a new issue for LoTW.

OK Clublog, think I might have noticed the issue when running the API push you need to feed it as follows

https://<url-to-clublog/index.php/clublog/upload/<cloudlog-user-name-that-has-clublog-info> the latter being the account you probably log in with

dwhacks, 2020-03-16:
Im not sure I understand, that seems like you want:
`https://clublog.org/index.php/clublog/upload/ve7dnt`

I use my callsign as the Cloudlog username, so whichever username it is that Im required for the API should be the same. I am able to login to the club log site with my callsign OR email.

The wiki here has a few mentions:

> http://<website-address>/index.php/clublog/upload/<username> replace with your Cloudlog 
 username,

this is [here](https://github.com/magicbug/Cloudlog/wiki/Clublog-Upload)

> ext, you can test it works by going to <url-to-cloudlog> /index.php/clublog/upload/<The-Cloudlog-Username-You-Just-Added-Logins-To> 

located [here](https://github.com/magicbug/Cloudlog/wiki/Setup-Club-Log-Integration)

and in the setup cronjobs section:

> 0 */6 * * * curl --silent https://<URL-To-Cloudlog>/index.php/clublog/upload/<username-with-clublog-login> &>/dev/null

The only other thing I can think of is using my e-mail as the club log login, but then I would have to allow @ in the URL string.

dwhacks, 2020-03-16:
I fixed the LOTW error, somehow there was a "space" after my password. Now it worked. This makes me even more confused as to why clublog doesn't work as this would make it seem like permissions are correct.

dwhacks, 2020-03-16:
I just had another thought, I am using the apache proxy/reverse proxy to point the incoming requests to the cloudlog VM. I have used this for Nextcloud running on my FreeNAS for a few years now without issues, but I did notice in the clublog config file there are some options for proxy, and I left them blank.

Could this have something to do with it?

magicbug, 2020-05-08:
If you run https://cloudlog.gfarc.ca/index.php/clublog/upload/VE7DNT the system is designed to return any errors Clublog returns, yes configuration of your server will play a big part but not something I can debug beyond testing code on my development and 80+ hosted live copies of Cloudlog.

closed 2020-05-10 by magicbug