magicbug/Cloudlog #438: Country not found

ClaudeMa, 2020-03-15:
**Describe the bug**
some days cloudlog stoped setting country name in new QSO 

updating DXCC retrun this

> Updating...
> Dxcc Entities: 402
> Dxcc Exceptions: 1050
> Dxcc Prefixes: 0

And check missin return

> 0 updated

update directory contains these files
[image]

Any idea?
Thank

[image]

dg9vh, 2020-03-15:
[image]
You should try with waiting till the update process is done... as I can see, you changed to another side while updating.

The process could take a minute depending on your internet-connection.

73 de Kim
DG9VH

magicbug, 2020-03-15:
Beat me too it, just leave it to run, speed will depend on hosting environment how quick it completes.

ClaudeMa, 2020-03-15:
Two days that i try, screenshot is take after 20mn waiting.

 "Done " is never show

retrying and retrying finished working maybe an  issue with clublog

Many thanks for help

closed 2020-03-15 by ClaudeMa