magicbug/Cloudlog #443: DXCC page stats not showing correctly

m0pcb, 2020-03-26:
On the new DXCC page the filter logic seems to be incorrect.

The 'show worked' checkbox should show everything that has been worked, including confirmed. However, it only shows worked but not confirmed.

I would also suggest the 'include deleted' should default to unchecked.

magicbug, 2020-03-26:
Interesting won't have time to check this till tomorrow but will flag it for @AndreasK79 as well.

Agree think deleted should be unchecked for most, as it will not affect most people.

AndreasK79, 2020-03-26:
That is how I designed it. For me, the whole point was to show those worked, but not confirmed. I was looking for a way to easily spot those not confirmed. I found this useful for boosting my confirmed dxcc totals. That's why I also implemented QSL / LoTW, You already know you worked it when it has a C for confirmed, so I don't see the point in just removing a C for a W. 

As for the deleted, for me it made sense to have it default as checked, since I have worked deleted dxcc's. Whether it's defaulting to on or off, isn't a big deal.

closed 2020-04-05 by magicbug