magicbug/Cloudlog #444: Main log entry window 'reset' button behaviour not correct

m0pcb, 2020-03-26:
When a call is entered into the Callsign box and left long enough for the lookup to happen and then the reset button is clicked the behaviour is not as expected. 

What I expect to happen is that the callsign field is wiped, along with the search results on the right panel and the DXCC name under the callsign box, essentially returning to a new 'live qso' page.

What actually happens is that the callsign box is wiped, but the search results persist, the DXCC name tag remains under the callsign field and the default RST S and RST R revert back to 59, even though it's still in CW mode, picked up from CloudlogCAT.

m0pcb, 2020-04-06:
> The way javascript handles is that once for example, a callsign inputted
> and a check is started, even if you hit reset the background process will
> still run hence the boxes populating I'm not a 100% certain the best way to
> resolve this actually as I cant see anything in the spec which might allow
> us to kill all the background calls

OK, it's not a big deal so it's not worth expending lots of energy to
resolve if it's not easy to do. :)

73, Iain AD5XI / M0PCB

magicbug, 2020-04-06:
Pah always worth expending a tiny bit of energy, fixed the Previous Contact box, when you reset the main list appears, just need to look at the CloudlogCAT resetting and the CW stuff

closed 2020-04-24 by magicbug