magicbug/Cloudlog #446: Some way to bulk update QSOs and track when exports have been done

m0pcb, 2020-03-26:
When exporting the log from cloudlog to then upload to LoTW via TQSL there is no way to batch update those QSOs in the log to show they've been uploaded to LoTW but not yet confirmed. 

The batch could be 100 QSOs for example, doing that manually isn't realistic.

Also, to make sure I only export QSOs that are not in LoTW already I have to manually cross check the latest uploaded QSO in LoTW before doing the Cloudlog export. A way of tracking at time of export the reason of export to then allow only QSOs not exported to be exported would be useful.

AndreasK79, 2020-03-29:
This sounds like it should be doable: 
1. Add another column to the table to indicate if exported. 
2. Add option to export page for exporting to LoTW (when checked, sets timestamp on exported QSOs to indicate LoTW-export)
3. Add button to mark either all QSOs as exported, or use date-interval if date(s) are chosen.

How does this sound? I can take a look at it if this approach seems ok.

m0pcb, 2020-03-29:
That sounds pretty much what Pete and I talked about as I wrote the
request. :-)

73, Iain AD5XI / M0PCB

magicbug, 2020-03-30:
@AndreasK79 Exactly what I suggested to @m0pcb if you can look into it happy to check it over, been bogged down with other bits going on

AndreasK79, 2020-03-31:
I'm thinking of the best way to do this, without too much confusion. Does this look ok @magicbug 
 and @m0pcb ?
[image]

magicbug, 2020-03-31:
That works for me personally, see what @m0pcb thinks :D

m0pcb, 2020-03-31:
Looks good to me :)

73, Iain AD5XI / M0PCB

commit 8fd62424 (2020-04-01): Merge pull request #452 from AndreasK79/LoTW_marking_on_export

Added possibility to mark QSOs as exported to LoTW. Fixes #446.
closed 2020-04-01 by magicbug