magicbug/Cloudlog #451: Old Import of data

AreaA41, 2020-03-31:
G'day..

After an absense of a couple of years, (bit of spare time on my hands atm.hi)
and after moving to a new host. Cloudlog stopped working properly.
My install could of been as old as two or more years...
Easiest Solution, Dump SQL, Install new instance import the historical contacts.
(bearing in mind, I couldn't do a polite export to adif...) 
After a bit of mucking about, I got things working and imported my old contacts. 
My historical contacts didn't show, but an 'insert of station ID' suggestion from anothe thread sorted that out - thank you.

I can now see all my old contact, but features, like worked squares map, dosen't show what squares I've worked.
I've run the Update Country Files, but that dosen't seem to do anything? 

any further thoughts ?

MarkA

magicbug, 2020-04-02:
Gridsquares rely on the data being available from logged contact, you can update the country files then recheck your log that would fill any missing dxcc info

AreaA41, 2020-04-04:
Still zero updates if I 'update the country flles'
The very long winded way around this, open the contact, save the contact.... Then the square appears.

magicbug, 2020-04-04:
Viewing a QSO will not update any data to diagnose what's going on will need to see some examples of data and screen shots

closed 2020-04-20 by magicbug