magicbug/Cloudlog #453: Cannot create station profile

janipewter, 2020-04-01:
Cannot create station profile because DXCC field is required but is not available.

Update Dxcc button in Admin -> Update country files also doesn't work.

Ubuntu 18.04.4
nginx 1.16.1
PHP 7.4.4

magicbug, 2020-04-01:
If the Update Country Files isn't working then you have an issue with your configuration, best bets check you have folder permissions correct, all the modules installed (see https://github.com/magicbug/Cloudlog/wiki/Installation) and check your local log files for errors.

The Update Country Files can take a while to run if you have a slow system.

janipewter, 2020-04-03:
OK thanks, it was a permissions issued. I changed ownership on the entire Clublog directory and that seems to have solved it.

closed 2020-04-03 by janipewter