magicbug/Cloudlog #454: after adding my 2nd profile with my new callsign (same person same country) awards doesn't show all dxcc contacts

AD7Djoe, 2020-04-02:
**Describe the bug**
After getting my official Finnish callsign OG7D, I have added a 2nd profile called OG7D.  My first profile being OH/AD7D.  

My active profile is now OG7D.  I have made about 5 or so contacts using this call.

Logging works fine.  And the logbook displays the contacts continuously on the page with correct details.

Now, since I am the same person operating in the same country, I am expecting that when I go to Awards -> when on page deselect Show confirmed and deselect Show not worked -> Show, my combined contacts for my 2 profiles will be shown.

The DXCC results are only showing for my latest OG7D contacts.

This is a similar situation with LOTW when adding a 2nd callsign for a person in the same country, the results need to be merged/combined.

**To Reproduce**

Steps to reproduce the behavior:
0. Have a profile and make several contacts, to Oman and France, e.g.
1. Add 2nd profile with new callsign.
2. Make 2nd profile active.
3. Log several contacts, to Germany and England, e.g.
4. Go to awards page.
5. Deselect Show confirmed and deselect Show not worked.
6. Click Show button.

** Results **

Germany
England

**Expected behavior**
Oman
France
Germany
England

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: windows 10
 - Browser [e.g. chrome, safari] chrome
 - Version [e.g. 22] latest

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

AndreasK79, 2020-04-06:
The way it is written right now, is that it is fetches info from the current profile. It is easy to adjust the queries to not care about the profile, but I'm not sure everyone wants this feature. I had a go at it locally, and  it was easy to adjust this and turn it on/off with a setting in the config file.

@magicbug should I make a PR with this feature?

magicbug, 2020-04-06:
I'd suggest getting the active station profile directly from the database that said might be better to update the session data when it's changed in station profile

AndreasK79, 2020-04-06:
I'm not sure I'm following you here. If I understand it correctly, he wants the DXCC award to count QSOs for both his profiles. How is this achieved with your suggestion?

The way I tested it locally, was to add an option in the cloudlog.php file. If this was true, I skipped checking on station_id,

magicbug, 2020-04-06:
In which case I miss understood, the charts should **only** show results for QSOs in the active station profile not the whole database.

magicbug, 2020-04-06:
Based on what I've seen this is working as designed thus not an issue nor a bug so I'll close the issue as I see no reason for awards to work in the requested format

closed 2020-04-06 by magicbug

AD7Djoe, 2020-04-07:
That is right, I want the profiles combined, that is the results combined.

Per dxcc rule number 10, the total of dxcc count can apply for a person using different calls in the same dxcc entity.

I have used OH/AD7D and now OG7D.

I want to see the combined total of the stations I have worked here in Finland.

Going back and forth is no fun between the profiles.

Br, Joe.

AD7Djoe, 2020-04-07:
To clarify my user scenario, I am currently using OG7D.  I hear a station from a callsign I don't remember operating, so as I am collect DXCC countries, I go to awards and see my listing.

Currently, the way it is implemented, I need to do this for OG7D, then, switch profiles to OH/AD7D then go to awards again to see if I have worked that country.

Perhaps having a checkbox on the page for "combined profile results" would be good.

What do you think ?